computer-lov/Nighttime-Parenting-Device #1: Set up repository on Github

computer-lov, 2022-02-02:
-[] Create repository
-[] Add collaborators
-[] Set up Milestones
-[] Set up Labels
-[] Associate with Project

BeatrizPer, 2022-02-02:
- [ ] sdas

BeatrizPer, 2022-02-02:
The bug will be used when something isn't working correctly and a fix needs to be found.
The design label will be related to prototyping and schematics. It will be added to 
 
documentation
Improvements or additions to documentation
 
easy
Less than 30 minutes
 
hard
More than an hour
 
hardware
Related to hardware
 
medium
Between 30 minutes and an hour
 
question
Further information is requested
 
software
Related to software and coding
 
testing
This will be used when preliminary or further testing is needing to be done on a task
 
wontfix
This will not be worked on

closed 2022-02-02 by arongoldberg

computer-lov, 2022-02-02:
@ffund Please review our project!