computer-lov/Nighttime-Parenting-Device #8: Confirm current consumption of OLED screen

BeatrizPer, 2022-02-04:
Testing will be done once part comes in to confirm current consumption of the OLED screen

arongoldberg, 2022-03-14:
I've confirmed that the OLED turns on via `i2cdetect -y 1` showing the expected address of 0x3d, and an indicator LED on the back lighting up:
[image]

I've run `pip3 install adafruit-circuitpython-ssd1327` to install the library for this display. I am trying to run [this](https://learn.adafruit.com/adafruit-grayscale-1-5-128x128-oled-display/circuitpython-wiring-and-usage#example-code-3076925-29) demo to see how much current is consumed when the screen is on.  When I try to run the code, I get a module not found error, and attempting to run `pip3 install adafruit-circuitpython-displayio` doesn't work either: 
[image]

There are some instructions [here](https://learn.adafruit.com/adafruit-grayscale-1-5-128x128-oled-display/circuitpython-wiring-and-usage#circuitpython-displayio-library-installation-3076879-8) that I was going to try but I don't really know what to download or where to copy it. After googling around, I don't think displayio is available without installing CircuitPython. I'm not too sure about that however, and I don't know what to do next in any case. @ffund can you please help me out?

arongoldberg, 2022-03-17:
@ffund Thank you, I installed those packages which allowed me to successfully run the example program. However, the screen remains off when I run the code. Here is a picture of what my AD2 picks up when I run the program: 
[image]
Reading the [datasheet](https://www.generationrobots.com/media/Ecran_OLED_0_96/SSD1327_datasheet.pdf) command table (sections 9 and 10), it looks like the first command, AE, turns the display off, the next bunch of commands configure various things, and the third to last one, AF, should turn the display on. Then a bunch of data is sent to GDDRAM to be displayed: 
[image]
I believe at this point the display should be on and there should be something on it, however it is still off.

ffund, 2022-03-21:
@arongoldberg Perhaps try the Luma library? https://github.com/rm-hull/luma.oled

arongoldberg, 2022-03-27:
@ffund I've tried to run the basic Luma test as well as some examples and they all result in a blank screen. Should we switch to a different OLED display?

ffund, 2022-03-28:
Sure @arongoldberg you can use the SSD1306-based display from DP1 if you prefer.

ffund, 2022-04-04:
@arongoldberg I had another of those 128x128 grayscale OLEDs around, and I tested it. I was able to run

```
sudo pip3 install luma.oled
git clone https://github.com/rm-hull/luma.examples
cd luma.examples/
python3 clock.py --display ssd1327 --height 128 --width 128 --i2c-address 0x3D
```

and get this:

[image]

If you want to try with the 128x128 OLED I used (in case you got a lemon), you can. Or, you can go with the SSD1306, your call.

arongoldberg, 2022-04-06:
3.6mA static text at default contrast, 3.9-4.8mA with running clock at max contrast. This is on the SSD1306 0.96" OLED from the fall semester, as even the confirmed working 1.5" SSD1327 OLED did not work with my Pi.

closed 2022-04-06 by arongoldberg

arongoldberg, 2022-04-06:
Reviewed and moved to done