computer-lov/Nighttime-Parenting-Device #9: Confirm current consumption of op-amp circuit

BeatrizPer, 2022-02-04:
Testing will be done once part comes in to confirm current consumption of the op-amp circuit

computer-lov, 2022-03-08:
The current consumption when the microphone was powered off was measured to be 0.03 mA.

[image]

The expected result was 0.0 mA. Therefore, the measured current is in agreement with the theoretical current.

The theoretical result was determine from this datasheet.

https://www.cuidevices.com/product/resource/cma-4544pf-w.pdf

computer-lov, 2022-03-08:
The current consumption when the microphone is powered on was measured to be 332 uA. 

[image]

The absolute maximum rating for the current consumption in this part is 500 uA. Therefore, the theoretical result is in agreement with the measured result. 

The theoretical result was determined from this datasheet.

https://www.cuidevices.com/product/resource/cma-4544pf-w.pdf

computer-lov, 2022-03-08:
The current consumption when the op-amp is powered off was measured to be 0.0 mA.

[image]

The specified value was also 0.0 mA. Therefore, the theoretical result is in agreement with the measured result.

The theoretical result was determined from this datasheet.

https://www.ti.com/lit/ds/symlink/opa344.pdf?HQS=dis-mous-null-mousermode-dsf-pf-null-wwe&ts=1646758941572&ref_url=https%253A%252F%252Fwww.google.com%252F

computer-lov, 2022-03-08:
The current consumption when the op-amp was powered on was measured to be 330 uA. The current was measured by removing the op-amp and then placing the positive end of the multimeter on the wire connecting to the power supply and the negative-end to the wire connecting to the positive terminal of the op-amp. 

[image]

closed 2022-03-09 by computer-lov

computer-lov, 2022-03-30:
@ffund we are considering moving to done. Does the current look reasonable?

ffund, 2022-03-30:
@computer-lov yes, seems OK.