computer-lov/Nighttime-Parenting-Device #10: Write Background of the report

computer-lov, 2022-02-07:
- [ ] Base background off of report from last semester
- [ ] Write in full paragraphs

computer-lov, 2022-02-16:
Beatriz completed the background of the report.

closed 2022-02-16 by computer-lov

BeatrizPer, 2022-03-01:
Night Time Parenting’s motivation and ultimate goal is to provide a solution to help ease the stressful and negative feelings that come along with taking care of an infant at night. After researching the endless struggles that new parents have to deal with when bringing in a new infant to their lives, it became evident that parents need to feel positive about caring for newborns so that they don’t get burnt out because then, the baby and parents suffer. Our company has worked on innovating an engineering design solution that enforces encouragement, peace, and equality within the parenting and child caring dynamic. Existing alternatives to solving this problem include meditation apps that help a parent follow a breathing routine to recompress. Motivational posters can be used to aid in encouragement. However, there hasn't been a system that fully integrates all these solutions into one to provide the optimal experience for the parent. Our company has taken an engineering approach to design and develop a fully integrated solution to meet the needs of parents who just need an extra helping hand.

BeatrizPer, 2022-03-02:
Reviewed and moved to done