computer-lov/Nighttime-Parenting-Device #15: Write UX section of report

arongoldberg, 2022-02-08:


BeatrizPer, 2022-04-08:
starting on UX section of report, will wait to finish after feedback is given by professor on issue #56 #60 #61

BeatrizPer, 2022-04-08:
Before creating the personas, a foundation was made to get a better understanding of who the user was. To do this, the team found various user stories on parents discussing their own experiences with nighttime care. After finding all these user stories sharing various sentiments towards the issue, a big picture was created to visualize it all. The team did this with an affinity map, which is a tool commonly used in UX research to help better organize ideas and data. All the user stories were added on sticky notes and scattered around the board. After all of them were collected and added, they were each organized into main categories. The team found three major categories relating some user stories with others. 

To be continued

BeatrizPer, 2022-04-12:
Changed label to hard since it should be substantial and require more than an hour to write

BeatrizPer, 2022-04-24:
Adding on to the UX design section
Beginning to connect our wireframes to the personas and basic UX principles discussed in class

BeatrizPer, 2022-04-24:
Wireframes and sketch

The browser based UI and the physical UI wireframes for our nighttime care device were built with the Balsamiq software. They can be seen in more detail in [Appendix B](https://docs.google.com/document/d/1wr1sxa2GU3sufivQdECUX-8X3vbnlJJqDqvwfQ-53co/edit#heading=h.hd28f0no7u1x). The physical wireframe is composed of many individual parts in order to successfully monitor the baby, detect stress in the caregiver, and activate routine to calm the caregiver. The main components for these functionalities include an OLED screen, an LED bar, a stereo decoder, a heart rate sensor, and a microphone. These are strategically placed around the wrist area to mimic a wearable watch. 

Our physical user interface consists of two dials placed right by the OLED. The dials are both clearly marked on their functionalities and are controlled as expected, which is turning left to decrease the value and turning right to increase the value. The first dial on the top is for brightness and the dial on the bottom is for volume. Therefore, turning the top dial to the right will result in an increase of brightness to the display. Finally, there is an SOS button further up the sleeve. If pressed, it will send a “help needed” message to the other designated caregiver. These three components are the only parts that the user will have to configure on the physical product when using it. Our goal with this design and choice of buttons is to reduce the performance load on the user and balance out the concept of flexibility-usability. Since our personas aren’t all tech savvy, such as Mei, reducing the cognitive and kinematic load were important for our product. This is why we chose to just use three buttons/dials as they are already intuitive. People know how to use dials and buttons. Additionally, they are all clearly labeled and easy to read. Most of our personas are already not having a fun time at night when they’re using the sleeve. For example, Liz gets very irritable when taking care of the baby at night. It would be a major inconvenience to add on a unintuitive product that could cause more points of frustration for her.

The volume, brightness, and SOS alert all are provided with feedback when they are adjusted and add onto the visibility principle of our product. When the volume is lowered, the music can be heard on a lower volume. When the brightness is lowered, the brightness is seen lowered on the screen. When the SOS button is pressed, the OLED displays a message confirming that a message was successfully sent.

 For the browser based UI, setup configurations can be done here before putting on the device. Because many of our personas consist of two caregivers such as Eli and George, and Clarence and Jessica, we added a setting to select which parent is setting up the configuration. This enables different saved settings and preferences for two different users. The analytics are tracked and plotted in the “Weekly Stats” page. This section, with parent division in particular, can be helpful for Liz and Jay as there is an issue evenly dividing the work amongst them. Each configuration has a constraint on the setting so users are able to understand what actions are possible. For example, the breathing setting has an on/off button. There is no inbetween which helps the user understand what is possible to adjust. 

Overall, the physical and browser based user interfaces are configured with respect to the personas created as well as the main user principles to provide an easy to use and well developed device.

BeatrizPer, 2022-04-27:
written, will do final review when report is done

closed 2022-04-27 by BeatrizPer