computer-lov/Nighttime-Parenting-Device #22: Write perfboard prototype section of report

arongoldberg, 2022-02-08:


BeatrizPer, 2022-05-06:
The prototype wiring is planned out before the soldering for the system is done. The perfboard layout planning sheet can be seen here [Appendix G - Perfboard Planning Sheet](https://docs.google.com/document/d/1wr1sxa2GU3sufivQdECUX-8X3vbnlJJqDqvwfQ-53co/edit#heading=h.bne4auqprwim). Most of the perfboard was dependent on the physical UI placement. The OLED, LED Bar, Dials, and heart rate sensor had a definitive location on the sleeve, which were to follow the physical UI placement discussed before. These were placed as priority in their designated locations. After these were soldered on, the perfboard was analyzed to find common grounds or powers. These commons were all connected and then finally connected by one wire to the pi. This helped prevent multiple ground/power cables going to the Pi and instead centralized them to a less populated area on the perfboard. 
     The perfboard was advertised as flexible so that was the original plan for the design. We’d flex the board and turn it into a wearable cuff. However, the perfboard cracked when attempting to flex it so instead, three different sections were cut and placed on a cardboard cuff for the prototype. One perfboard was for the top of the cuff, which is what would include the user interactive pieces such as the LED Bar, Dials, OLED, microphone, and SOS button. The bottom of the cuff would have another section of the perfboard which would have all the long wires coming out of it so this is where the Raspberry Pi Zero and Stereo Decoder are placed. Additionally, the heart rate sensor needs a reading from the wrist so a third perfboard was made to include the heart rate sensor. The prototype is seen here.
	Besides that, cables were used when going through long distances. The convention of red being power, green/black being ground, and other colored cables being signal was followed. The rest was about strategic placement. For example, the LED bar needs to be connected to many pins on the shift register. I oriented the bar and shift register so the two sides that had to be soldered together were already close by. This made connecting them so much easier.

BeatrizPer, 2022-05-06:
[image]

closed 2022-05-06 by BeatrizPer