computer-lov/Nighttime-Parenting-Device #24: Write project planning section of report

arongoldberg, 2022-02-08:
Include weekly screenshots of project board

BeatrizPer, 2022-04-08:
Starting to write project planning section of report

BeatrizPer, 2022-04-08:
The project work was split by three members. Github and its project board are the main sources that have contributed to effective and efficient collaboration between the team. With the Github Project, the team was able to assign, organize, document, and discuss issues when they arose on all the tasks at hand. Before creating the tasks, the whole project was split up into various milestones. These milestones included the following: preliminary planning, learning about each individual component, UX design, application logic, circuit prototyping on PCB and Perfboard, browser based UI, and finally packaging and presentation. By setting up these milestones, they enabled the team to have a soft deadline to stay on track and have a general sense of direction on when individual tasks should be completed. 

After these milestones were established, the tasks were created and assigned to members. Assigning these tasks created a sense of accountability for the work. Everyone knew what they were responsible for and what had to be delivered. In Github, the team commented on their progress on each task in the comment section. This allowed easy hand-offs of work in case one member had to be given a task that another member started. For example, Andrew was having difficulties with the affinity map and persona creations so Beatriz took over. She was able to go through Andrew’s documentation and pick off where he left off. Andrew took over Beatriz’s work with the heart rate sensor. Andrew was able to figure out the connections of the breakout board with a screenshot Beatriz had added in the comment section of her task without even having to ask her. These things all contributed towards effective communication throughout the project. Slack is another key component to functional communication. Whenever the team had any questions or needed feedback on current work, they would ask the slack channel. This is where other team members would provide feedback/answers. For example, after Beatriz finished the affinity map she sent the link to the Slack. This is where Aron noticed an issue with one of the categories on the map and suggested an improvement. 

To stay on track, another key factor was taking weekly notes. The following template was set up and used during every meeting. This helped the team stay consistent and organized through every meeting by following the same order of things every time. 

[image]

 
Overall, with the combination of Github and its project board, Slack, and weekly meetings, the team was able to effectively collaborate, establish goals, plan tasks, and meet all the objectives in a timely manner.

BeatrizPer, 2022-04-08:
Moving to review

BeatrizPer, 2022-04-08:
**need to include weekly project board screenshots**

closed 2022-04-08 by BeatrizPer