computer-lov/Nighttime-Parenting-Device #40: Check part "liveness" for Stereo Decoder

arongoldberg, 2022-02-09:
- [ ] photo/schematic
- [ ] code and/or instructions
- [ ] expected result of "liveness" test

arongoldberg, 2022-02-18:
I've been having trouble getting this to work. I followed [this](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-wiring) schematic to wire it up and attached pictures.
[image]
[image]
First, I used [this](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-usage#fast-install-2712950-3) "fast install" tutorial from adafruit, and no sound came out of my headphones (which were powered) during the test of re-running the script after reboot. I tried both enabling and disabling the `/dev/zero playback service` as suggested in the tutorial. 

Then, I tried doing [this](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-usage#detailed-install-2712959-16) "detailed install". After the detailed install, I tried [this test](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-test#simple-white-noise-speaker-test-2711201-1) and [this test](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-test#simple-wav-speaker-test-2711201-1) and didn't hear any sound with two different (confirmed working) pairs of headphones, one powered and one not. This is what my screen displayed during the tests:
[image]

Also, if I moved the decoder around, I sometimes got static coming through the headphones. This happened with both headphones, both while performing the audio test and while nothing was running.

@ffund what should I try next?

ffund, 2022-02-21:
thanks for the update, here are some suggestions:

* the photo of your wiring doesn't show any obvious problems
* you could post more details of the manual install part (e.g. screenshots) to see if anything is wrong there. You can also show the output of `lsmod` and `gpio readall` after the reboot. (`lsmod` should show that the correct kernel modules are loaded, and `gpio readall` should show that the pins are in the correct state for I2S. You can see at [pinout.xyz](https://pinout.xyz/pinout/pin35_gpio19) that the three PCM pins should be in ALT0 mode.)
* the "if I moved the decoder around, I sometimes got static coming through the headphones" suggests a poor electrical connection - maybe a bad solder joint on the Pi header, or on the UDA1334A header. You could connect these through a breadboard instead of directly, then you could connect your AD2 scope to each signal in turn, and see what's going on there.

arongoldberg, 2022-02-23:
Thanks for the suggestions. Here is the output of `gpio readall` and `lsmod`:
[image]
[image]
The three PCM pins are in ALT0, and I'm not really sure what to look for from `lsmod`.
Here are the two non-empty files from the manual install and the links to their parts in the install:
[First](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-usage#create-asound-dot-conf-file-2712971-28)
[image]
[Second](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-usage#add-device-tree-overlay-2712977-34)
[image]
Note that in the second one, the line with `dtparam=i2s=on` is commented out. This is what is shown in the screenshot on the install page. I've tried both with and without commenting that line out.

I connected everything through my breadboard and used my AD2 to try and analyze the I2S signal. It appears that nothing is coming over the data line. I've tried different wires and different breadboard rows with the same result. The oscilloscope is connected to the data line in this picture.
[image]
When I connected the oscilloscope to the word select and bit clock line, they acted as expected, oscillating as seen in the logic portion of the screenshot above.

@ffund what should I try next?

ffund, 2022-02-23:
Re: the `lsmod` output - what we were looking for was:

* `snd_bcm2835` module is not loaded (corresponds to the "Disable headphone audio (if it's set)" section of the manual install)
* the relevant PCM module should be allowed to load (corresponds to the "Update /etc/modprobe.d (if it exists)" section of the manual install). It's not clear which module is specifically relevant to the Pi Zero, though.

In the short term, I think the best next step would be to get together with the other student who has an I2S audio device. It's the same install process. You can try using his I2S device (which is working) on your Pi, and he can try using yours on his Pi - that will help us narrow down the issue to "bad breakout board" or "bad electrical connection on Pi" or "bad install".

arongoldberg, 2022-02-24:
Today in the lab I tried my decoder with another student's Pi and it worked correctly. I then tried my decoder with my Pi and his SD card in it, which did not work. This points to two possibilities: either one of the PCM pins on my Pi was loose, or there was something wrong with my install of the I2S software on my Pi.

arongoldberg, 2022-02-25:
After having the pins on my Pi resoldered, I tried my decoder again. It didn't work again. I then tried my Pi's SD card in another Pi and this finally worked! I was able to successfully perform the [fast install](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-usage#fast-install-2712950-3) on this new Pi and its associated test, as well as [this test](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-test#simple-white-noise-speaker-test-2711201-1) and [this test](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-test#simple-wav-speaker-test-2711201-1). Then, I swapped the SD card back into my Pi and that worked too! I was even able to do the fast install from home, so it seems like my network issues were solved as well.

arongoldberg, 2022-02-25:
I'm closing this issue as I successfully determined that this decoder works. I wired it up as follows:
[image]
I followed [these instructions](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-usage#fast-install-2712950-3) and performed the test at the end of the install of a voice alternating between saying "front left" and "front right," with the sound coming out of the corresponding side of the headphones. I also performed [this](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-test#simple-white-noise-speaker-test-2711201-1) white noise test, [this](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-test#simple-wav-speaker-test-2711201-1) WAV file test of someone speaking the words "front center", and [this](https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-test#simple-mp3-speaker-test-2711201-1) MP3 stream test.

closed 2022-02-25 by arongoldberg

BeatrizPer, 2022-03-02:
Working and sent to done