computer-lov/Nighttime-Parenting-Device #42: Project Specific Tests: Heart Rate Sensor

computer-lov, 2022-02-09:
- [ ] describes and justify tests
- [ ] explain how to do them
- [ ] explain what results were

BeatrizPer, 2022-02-23:
temperature test is written below
```
    def getTemp(self):
        # temperature read in integer and fraction, both added
        # to find total temp
        
        # Enabling the temperature reading
        i2c.write_byte_data(HR_ADDR, 0x21, 1) # TEMP_EN
        i2c.write_byte_data(HR_ADDR, 0x03, 1) #DIE_TEMP_RDY_EN
        
        # Address for temp int = 0x1F
        T_int = i2c.read_byte_data(HR_ADDR, 0x1F)
        # Address for temp frac = 0x20
        T_frac = i2c.read_byte_data(HR_ADDR, 0x20)
        
        temp = T_int + T_frac
        return temp
```

BeatrizPer, 2022-03-01:
The testing code function (main.py) is responsible for calling the heart rate monitor and starting up the sensor to begin the readings. After cloning the repository, the following code is ran to execute the code on the board.

[image]

The MAX30102 should turn on, ready to get its first reading. This is indicated by the red LED light on the finger pad sensor seen above.
After the light has turned on, a finger can be placed right on the light as shown in the picture below. 
[image]

After the finger has been on the sensor for a few seconds, the terminal begins to print out BPM and SPO2 readings. The following figure below shows a few readings returned after I placed my finger on the sensor as explained above.
Although these readings appear to be accurate, there were a few outliers that were returned that were ignored. With these readings, the MAX30102 Heart Rate Sensor has successfully been set up and works!

[image]

closed 2022-03-01 by BeatrizPer