computer-lov/Nighttime-Parenting-Device #50: Project Specific Tests: LED Bar

computer-lov, 2022-02-09:
- [ ] describes and justify tests
- [ ] explain how to do them
- [ ] explain what results were

BeatrizPer, 2022-03-02:
- turn bars on for 4 seconds, hold, then turn off for 4 seconds

BeatrizPer, 2022-03-06:
wrote code to turn LED Bar on, wiring seen below

[image]

BeatrizPer, 2022-03-06:
import spidev
import RPi.GPIO as GPIO
import time
from time import sleep

spi = spidev.SpiDev()
dt = 0.4

def setup():
        GPIO.setwarnings(False)
        GPIO.setmode(GPIO.BCM)
        GPIO.setup(14, GPIO.OUT)
        GPIO.setup(15, GPIO.OUT)
        spi.open(0,0)    # SPI Port 0, Chip Select 0
        spi.mode = 0b00
        spi.max_speed_hz = 7629

def breathe_in():
        spi.xfer([0b00000000])
        GPIO.output(14, GPIO.LOW)
        GPIO.output(15, GPIO.LOW)
        sleep(2)
        spi.xfer([0b00000001])
        sleep(dt)
        spi.xfer([0b00000011])
        sleep(dt)
        spi.xfer([0b00000111])
        sleep(dt)
        spi.xfer([0b00001111])
        sleep(dt)
        spi.xfer([0b00011111])
        sleep(dt)
        spi.xfer([0b00111111])
        sleep(dt)
        spi.xfer([0b01111111])
        sleep(dt)
        spi.xfer([0b11111111])
        sleep(dt)
        GPIO.output(15, GPIO.HIGH)
        sleep(dt)
        GPIO.output(14, GPIO.HIGH)

def breathe_out():
        sleep(2)
        GPIO.output(14, GPIO.LOW)
        sleep(dt)
        GPIO.output(15, GPIO.LOW)
        sleep(dt)
        spi.xfer([0b11111111])
        sleep(dt)
        spi.xfer([0b01111111])
        sleep(dt)
        spi.xfer([0b00111111])
        sleep(dt)
        spi.xfer([0b00011111])
        sleep(dt)
        spi.xfer([0b00001111])
        sleep(dt)
        spi.xfer([0b00000111])
        sleep(dt)
        spi.xfer([0b00000011])
        sleep(dt)
        spi.xfer([0b00000001])
        sleep(dt)
        spi.xfer([0b00000000])
        sleep(dt)

while True:
        setup()
        breathe_in()
        breathe_out()

BeatrizPer, 2022-03-06:
breathe_in() lights up LEDs by sending highs to the designated LEDs in order to create a guide to follow while inhaling. There is a time delay in between each LED being turned on. Its used by calling it
breathe_out() turns off LEDs by sending lows to the designated LEDs in order to create a guide to follow while exhaling. It works by calling it.
setup() sets up SPI and sets extra LED bars to GPIO pins since shift register allows 8 to be connected.

BeatrizPer, 2022-03-06:
https://user-images.githubusercontent.com/98904988/156908646-37dd75b4-8da6-4936-a8ca-e77b1855348b.MOV

BeatrizPer, 2022-03-06:
Project specific tests are done

closed 2022-03-06 by BeatrizPer

BeatrizPer, 2022-03-09:
Reviewed and done