computer-lov/Nighttime-Parenting-Device #51: Check part "liveness" for Heart Rate Sensor

arongoldberg, 2022-02-09:
- [ ] photo/schematic
- [ ] code and/or instructions
- [ ] expected result of "liveness" test

BeatrizPer, 2022-02-21:
Checked that its powered on and successfully detected through I2C, added pictures/code/confirmation to report

BeatrizPer, 2022-02-23:
Reopening issue to comment progress

BeatrizPer, 2022-02-23:
[image]

BeatrizPer, 2022-02-23:
To connect the heart rate sensor and get it working, you will need the following parts.

- Heart Rate Sensor for Max 30102
- Raspberry Pi
- Jumper cables

This heart rate sensor breakout board can accept a voltage range including 3.3V, therefore the power given directly from the raspberry pi is within range to power the sensor. Because of this, no additional component is needed for power. Since the breakout board has a built-in ADC, the values are all readable. The low level schematic for the connections are seen in the figure below.

Following the image in figure 1 above, to connect the part you must first give it power. Connect the 3.3V pin on the top left of the Pi to the pin on the Max 30102 board that reads “Vin”. Then, connect a ground pin from the pi (using the bottom left in the image) and connect this with the “GND” pin on the board. Finally, since we will be using i2c to communicate, the pins for these must be wired. With another jumper cable, the GPIO 2 pin “SDA” on the pi is connected to the “SDA” pin on the Max board. The GPIO 3 pin “SCL” is then connected to the “SCL” pin on the Max board. After these four connections have been successfully created, the Max 30102 board is all wired up and ready for testing.

BeatrizPer, 2022-02-23:
**Testing I2C Connection**
After the board is connected, the connection on I2C must be checked to verify that the Pi is reading the device properly. The steps are as follows which reference the I2C lab instructions. 

I2C must first be enabled on the Raspberry Pi OS. After logging onto the Pi, run the following command

`sudo raspi-config`

The following window will pop up as seen in figure 2. Navigate to interfacing options, select I2C and click OK.

[image]

After this is done, the following window will appear confirming that you have configured the interface correctly.

[image]

To verify in the program that it is enabled, run the following command.

`lsmod | grep i2c`

The following in figure 4 will be returned on the screen if the setup was successful. 

[image]

It is now time to confirm that the Max 30102 board is being read and is addressable. This will be done by scanning the entire I2C address space and looking for responses from each address. Run the following command in the terminal to do this.

`i2cdetect -y 1`

The result should be the peripheral address of the Max board. This expected value will be found in the datasheet of the part. The peripheral address is seen below in figure 5.

[image]

The peripheral address for the Max board is 0b1010111, which in hex is equal to 0x57. Running the code from above, the expected result will be this device address present in the I2C address space. Looking at the I2C response in the terminal after running the command, the following is seen.

[image]

As expected, the peripheral address for our board is seen at 0x57 which is what it should be according to the datasheet. This confirms that our part is successfully connected through I2C and ready to read values.

closed 2022-02-23 by BeatrizPer

BeatrizPer, 2022-03-02:
working and done