computer-lov/Nighttime-Parenting-Device #60: Wireframe for browser-based UI

arongoldberg, 2022-03-02:


BeatrizPer, 2022-03-23:
still need to add in browser a section to send message to oled display

BeatrizPer, 2022-03-29:
current browser design
[image]

BeatrizPer, 2022-03-30:
Removing "My average heart beat is"

and including other pages on user interface

[image]

[image]

[image]

closed 2022-03-30 by BeatrizPer

BeatrizPer, 2022-04-20:
Moving to done