computer-lov/Nighttime-Parenting-Device #61: Wireframe for physical product UI

arongoldberg, 2022-03-02:


BeatrizPer, 2022-03-29:
Adjusting Wrireframe based on feedback

- add SOS button to call for other caretaker
- change LED screen to also show wakeup time, current time, bpm
- change buttons to dials so easily and quickly change setting for brightness/noise

BeatrizPer, 2022-03-29:
Asking team for feedback

computer-lov, 2022-03-30:
Reviewed and moving to done.

closed 2022-03-30 by computer-lov

BeatrizPer, 2022-04-04:
@ffund 
Can you review our updated UI after making adjustments from your feedback

ffund, 2022-04-13:
@BeatrizPer 

We discussed the option of changing to a "lights on/off" button instead of a brightness dial. In that case, what is the layout of the "lights" button vs "SoS" button - where do those both go? (you want to make sure that it's not easy to accidentally trigger the SoS when user meant to turn light off".

Actually, it seems to me that the user might want to adjust brightness of the LED bar graph. Perhaps you should keep the brightness dial for adjusting the LED, and have it to turn display off if the brightness dial is set at the far (low) end.

BeatrizPer, 2022-05-10:
Updated wireframe
[image]