computer-lov/Nighttime-Parenting-Device #62: Create affinity map

computer-lov, 2022-03-02:
- [ ] create affinity map like in class example

computer-lov, 2022-03-08:
Created an affinity map from the compiled user research using Jamboard.

https://jamboard.google.com/d/1haQbWqZEbhgPHvbV2ly60fo2-vuMrqLiXxJ_AHvrjAY/viewer?f=0

computer-lov, 2022-03-09:
Moving to done.

computer-lov, 2022-03-30:
Reassigned to Beatriz

BeatrizPer, 2022-04-01:
Working on getting user stories to be centered towards parenting and not on existing devices

BeatrizPer, 2022-04-04:
Asked for feedback from Andrew and Aron on affinity map created 
[image]

BeatrizPer, 2022-04-04:
[image]
Will update night time care causes guilt part

BeatrizPer, 2022-04-04:
Added new stories and added the personas on scale
[image]

BeatrizPer, 2022-04-04:
@ffund Can you review our affinity map/personas?
https://jamboard.google.com/d/1haQbWqZEbhgPHvbV2ly60fo2-vuMrqLiXxJ_AHvrjAY/edit?usp=sharing

BeatrizPer, 2022-04-15:
Adjusting affinity map per Professor feedback

- Find more user stories where people are more chill about night time parenting for the left extreme
- Change guilt section to be more specific - guilty about what? (Now centering towards sleep training and emotions to that)

BeatrizPer, 2022-04-15:
[image]
updated

closed 2022-04-15 by BeatrizPer