computer-lov/Nighttime-Parenting-Device #63: Create personas

computer-lov, 2022-03-02:
- create at lest n+1 for n-sized team (at least 4 for us)

computer-lov, 2022-03-02:
- [ ] Persona 1 (Andrew)
- [ ] Persona 2 (Andrew)
- [ ] Persona 3 (Beatriz)
- [ ] Persona 4 (Aron)

computer-lov, 2022-03-09:
Created two personas: Jessica/Clarence and Beth. Here is the link to the Jamboard.

https://jamboard.google.com/d/1haQbWqZEbhgPHvbV2ly60fo2-vuMrqLiXxJ_AHvrjAY/viewer?f=2

computer-lov, 2022-03-09:
New working link:

https://jamboard.google.com/d/1haQbWqZEbhgPHvbV2ly60fo2-vuMrqLiXxJ_AHvrjAY/edit?usp=sharing

computer-lov, 2022-03-21:
Added two more personas:

https://jamboard.google.com/d/1haQbWqZEbhgPHvbV2ly60fo2-vuMrqLiXxJ_AHvrjAY/edit?usp=sharing

BeatrizPer, 2022-03-29:
Feedback from professor:
follow format given to really show different characteristics of the persona

BeatrizPer, 2022-03-30:
Made changes to Marc Persona to reflect template and add context/pain point/goals/ etc

[image]

BeatrizPer, 2022-03-30:
added liz & Jay
[image]

BeatrizPer, 2022-03-30:
Need to also fix Jam Board since personas arent the only issue

BeatrizPer, 2022-04-04:
Changed Clarence + Jessicas Personas
[image]

BeatrizPer, 2022-04-04:
Added Eli and George

[image]

BeatrizPer, 2022-04-04:
n

BeatrizPer, 2022-04-15:
Adjusting personas based on Professor feedback

- Tell more about night time care, who is taking care of baby and how often?

BeatrizPer, 2022-04-19:
Updated Marc:

[image]

BeatrizPer, 2022-04-19:
Updated Eli and George

[image]

BeatrizPer, 2022-04-19:
Updated Jessica and Clarence:

[image]

Updated Liz and Jay:
[image]

Updated Mei:
[image]